analogdevicesinc/libsmu #176: Question on how to read DIO inputs in Python 3

damercer, 2021-01-27:
Hi:
I have a question / issue on how to read the state of the digital inputs in Python 3 vs Python 2.
In python 2 This works:

Dval = devx.ctrl_transfer( 0xc0, 0x91, 4, 0, 0, 1, 100)
print(Dval[0])
if Dval[0] == 1:
....

The same code in Python 3 returns an error:

TypeError: 'map' object is not subscriptable

 Printing just Dval outputs this:

<map object at 0x0764F630>

So:
1. how do I get this working in Python 3
2. how do I access the contents of a "map" object.

Whatever changed in the Python bindings is not backward compatible with Python 2?

There isn't an example in the Python Bindings folder in the repo explaining how to read the digital pins?

Thanks

Doug

damercer, 2021-01-27:
Further to this question:
In python 2, Dval = devx.ctrl_transfer( 0xc0, 0x91, 4, 0, 0, 1, 100) returns a list type.

But in Python 3 using list(devx.ctrl_transfer( 0xc0, 0x91, 4, 0, 0, 1, 100)) to convert map type to list type returns::

Dval = list(devx.ctrl_transfer( 0xc0, 0x91, 4, 0, 0, 1, 100))
TypeError: ord() expected string of length 1, but int found

So apparently using the list function to covert map type to list type in this case at least is not a solution.

So What's the answer?

cristi-iacob, 2021-01-27:
Hi @damercer!

You are right, there is a problem with the map compatibility between Python2 and Python3. We found a solution and already merged it into master. After this modification, the Python3 ctrl_transfer() method will return a list as well (like in Python2).

You can rebuild libsmu/pysmu in order to get the modification. Right now we are facing an issue with Appveyor, so the Windows installers will only be ready in the next days.

Please let me know if you have any further questions. Thank you!

cristi-iacob, 2021-01-27:
Until we merge the Appveyor modifications into master, you can try using the Windows installers from the [development branch](https://ci.appveyor.com/project/cristi-iacob/libsmu/builds/37476192/job/8oqw9uniqg3japai/artifacts).

damercer, 2021-01-27:
Thanks for the quick response.
I think most users of ALICE use my Python 2.7 built Windows package and this issue only really impacts users who might be using the external analog Multiplexer features running Py 3 from source code.

I can wait for a release from Master for the next official release of ALICE.

Doug

closed 2022-02-11 by AlexandraTrifan