analogdevicesinc/libsmu #181: Ending continuous mode with session.end() not working

jmball, 2021-03-29:
I'm finding that calling session.end() after doing a write-read cycle in continuous mode always hangs indefinitely. This is a minimal example that shows the problem:

```
import pysmu

session = pysmu.Session()
dev = session.devices[0]

chan_a = dev.channels['A']
chan_b = dev.channels['B']
chan_a.mode = pysmu.Mode.SVMI
chan_b.mode = pysmu.Mode.HI_Z

num_samples = 1

print("Starting continuous mode")
session.start(0)

chan_a.write([1] * num_samples, -1)
data = chan_a.read(num_samples, -1)
print(data)

print("Ending continuous mode...")
session.end()
print("Continuous mode ended")
```

The program hangs after `print("Ending continuous mode...")`. Is this a bug or am I doing something wrong?

This may be related to #109. I asked the same question in that thread and the above script worked as expected for @damercer. I'm running python v3.7.10 and libsmu v1.0.3 (installed from conda-forge) on Windows 10. Could there be a problem with this configuration?

cristi-iacob, 2021-03-30:
Hi @jmball! 

Thank you for the feedback! Your script seems correct and this is the proper way to use a continuous session. Although it seems that not the session.end() call is the problem here. I tried to remove that line and the script still hangs, but in a little different style: with the session.end() I can't even stop the program using CTRL+C; without the session.end() call I can stop the program with CTRL+C. Is this the case for you as well?

As a side note, I tried to run the script on Ubuntu and it works as expected. So I suspect the issue is somehow Windows specific. I will investigate this issue further and will come back to you with an answer as soon as possible. 

Thank you, 
Cristian

damercer, 2021-03-30:
As I said in the response to #109 the script as written works as expected on Win 10 running from the Python 3.7 IDLE. So this might be (probably is?) related to the Conda package.

Doug

jmball, 2021-03-30:
Hi @cristi-iacob, thanks for looking into it! Yes, I see the same thing. With session.end() I can't stop it with CTRL+C, without session.end() I can.

hehonglu123, 2021-12-09:
Is there any update on this issue? I had it working for a while, but encountered the problem when setting up new computers with exactly same settings (libsmu1.0.3+python3.7 IDLE). The working computer has Win11 on it now, was Win10 during the first installation.

AlexandraTrifan, 2022-02-11:
Hi,

We managed to track down this issue.
You mentioned you were using the Conda package for libsmu. The Conda package for the new release(1.0.4) will be done soon.
In the meantime, you can try the latest release binaries that we prepared. (https://github.com/analogdevicesinc/libsmu/releases/tag/v1.0.4).

This one **does not include the Python bindings** in the official release, so you just download the libsmu setup file and install the base library + the smu command line utility. 
As for the Python version, you can just use the following command, which does not require you to pre-download any .whl file (it just downloads it from the official Test PyPi repository).
You can install that one using pip, just be sure to upgrade your pip first.
`python.exe -m pip install --upgrade pip`
`pip install -i https://test.pypi.org/simple/ pysmu` 
This one will detect your Python version and choose the specific wheel(pip package) for it.

The hang at `session.end()` in continuous mode should not appear anymore, it was linked to some mutexes not being locked/unlocked properly.

Thank you!
-Alexandra