analogdevicesinc/libsmu #183: Which Windows installer fixes issue #176?

damercer, 2021-04-26:
H:
Which Windows installer fixes issue #176 from back in January? The "official" master release is from Sept 2020 which is way before the issue was identified.
I'm about to build the next release of ALICE for May and need it to work in Python 3 in Linux as well so a pointer to the correct Linux installer package would be helpful as well..
Thanks
Doug

AlexandraTrifan, 2021-04-26:
Hi,

The changes for the issue you mentioned were added on the master branch. There is no official release that includes those yet.
For Windows, the nightly build from Appveyor can be used: https://ci.appveyor.com/project/analogdevicesinc/libsmu/builds/37780266/job/bsmgh3ykjnpfl00t .
We need to migrate OSX and Linux builds to Azure Pipelines (hopefully in the next weeks) and then we'll have nightly builds for all the operating systems.

Before creating a new official release of libsmu, there are still some issues that require fixing and that we are currently investigating. 

-Alexandra

damercer, 2021-04-26:
I wanted to test the Windows version in case a user (are recommended to use released versions only) wanted to run from source code but I actually only absolutely need a fixed Linux version for a release. I guess I can compile from the master source (like in the past) in the mean time.

damercer, 2021-04-28:
OK, so since there isn't a Linux package yet with the fix, I'm trying to compile from source as I said above.
I'm using a newly created VM of Ubuntu 20.04.
I did all the install steps and verified that everything I needed was installed and up to date.
I cloned libsmu.
Did the build, compile, install steps from the READme and everything seemed to have completed without errors.
The smu cli was there and when I ran it, it said:
smu: error while loading shared libraries: libsmu.so.1: cannot open shared object file: no such file or directory

Same sort of error when I tried to import pysmu in Python.

Must be a path issue that was not set as part of the install process? Do I need to reboot or reset some configuration to get it to find the object file. Must be a step missing here?
Doug

AlexandraTrifan, 2021-04-28:
Hi,

Did you try to run "sudo ldconfig"?

-Alexandra

damercer, 2021-04-28:
The sudo ldconfig fixed it. I didn't see that step in the READme? I think it use to be there but not anymore? I go back and look more closely. At any rate it needs to be included right after the last step:
analog@analog:~$ sudo make install

Thanks
Doug

AlexandraTrifan, 2021-04-28:
It's written in the last section of the readme: Linux FAQ.
We can move it after the sudo make install step.

-Alexandra

damercer, 2021-04-28:
That will be helpful. That final step is likely to be forgotten if it is not right after the install step, I did anyway.
Anyway everything is working fine now and the digital read back is tested and fixed in Windows and Linux.
I can proceed with making the new May release of ALICE now.
Thanks
Doug

closed 2022-02-11 by AlexandraTrifan