analogdevicesinc/libsmu #187: ctrl_transfer() documentation/examples

jmball, 2021-06-23:
I've been looking into using the `ctrl_transfer()` method in pysmu to get lower level control of the device. However, I'm struggling to find any clear documentation or examples showing how to use it and what settings are possible with the ADALM1000. Are there any detailed resources available to learn about it?

AlexandraTrifan, 2021-06-23:
Hi,

Pysmu.ctrl_transfer() is based on libusb_control_transfer. You can check out the libusb documentation here: https://libusb.sourceforge.io/api-1.0/group__libusb__syncio.html#gadb11f7a761bd12fc77a07f4568d56f38 for parameter explanation. The link to the official libsmu documentation is this one: http://analogdevicesinc.github.io/libsmu/classsmu_1_1Device.html#ae0d035b18c9ae682c9cd651f94417274 .

For the request field for the setup packet (bRequest argument) all the options are specified in the firmware. 
https://github.com/analogdevicesinc/m1k-fw/blob/master/src/main.c#L614-L830
The highlighted lines contain all the options available for this low level USB access.

One simple example:
`import pysmu`
`s = pysmu.Session()`
`dev = s.devices[0]`
`dev.ctrl_transfer(0x40, 0x03, 7, 0, 0, 1, 100)`

This will write value "7" and all the 3 LEDs will light up, as explaned in the code comments: 
 https://github.com/analogdevicesinc/m1k-fw/blob/master/src/main.c#L650
 
 Please let me know if you have any others questions!
 
 Thanks!
 -Alexandra

jmball, 2021-06-25:
@AlexandraTrifan , thanks for sending the info!

closed 2022-02-11 by AlexandraTrifan