analogdevicesinc/libsmu #190: Cann't compile on Ubuntu 20.04

meschi, 2021-10-03:
I'm trying to compile with the python3 bindings calling cmake like this: cmake .. -DBUILD_PYTHON=ON -DUSE_PYTHON2=OFF
This is my output on make:
...
[ 44%] Building CXX object src/CMakeFiles/smu.dir/signal.cpp.o
[ 55%] Building CXX object src/CMakeFiles/smu.dir/usb.cpp.o
[ 66%] Linking CXX shared library libsmu.so
[ 66%] Built target smu
[ 77%] Generating timestamp
running build
running build_py
copying pysmu/__init__.py -> build/lib.linux-x86_64-2.7/pysmu
running build_ext
Traceback (most recent call last):
  File "/home/meschi/opt/libsmu/bindings/python/setup.py", line 275, in <module>
    'Programming Language :: Python :: 3.6',
  File "/usr/lib/python2.7/dist-packages/setuptools/__init__.py", line 145, in setup
    return distutils.core.setup(**attrs)
  File "/usr/lib/python2.7/distutils/core.py", line 151, in setup
    dist.run_commands()
  File "/usr/lib/python2.7/distutils/dist.py", line 953, in run_commands
    self.run_command(cmd)
  File "/usr/lib/python2.7/distutils/dist.py", line 972, in run_command
    cmd_obj.run()
  File "/usr/lib/python2.7/distutils/command/build.py", line 128, in run
    self.run_command(cmd_name)
  File "/usr/lib/python2.7/distutils/cmd.py", line 326, in run_command
    self.distribution.run_command(command)
  File "/usr/lib/python2.7/distutils/dist.py", line 972, in run_command
    cmd_obj.run()
  File "/home/meschi/opt/libsmu/bindings/python/setup.py", line 127, in run
    from Cython.Build import cythonize
ImportError: No module named Cython.Build
make[2]: *** [bindings/python/CMakeFiles/python.dir/build.make:62: bindings/python/timestamp] Error 1
make[1]: *** [CMakeFiles/Makefile2:163: bindings/python/CMakeFiles/python.dir/all] Error 2
make: *** [Makefile:130: all] Error 2

after apt-get remove python-setuptools i get:
...
[ 66%] Built target smu
[ 77%] Generating timestamp
Traceback (most recent call last):
  File "/home/meschi/opt/libsmu/bindings/python/setup.py", line 15, in <module>
    from setuptools import setup, find_packages, Extension, Command
ImportError: No module named setuptools
make[2]: *** [bindings/python/CMakeFiles/python.dir/build.make:62: bindings/python/timestamp] Error 1
make[1]: *** [CMakeFiles/Makefile2:163: bindings/python/CMakeFiles/python.dir/all] Error 2
make: *** [Makefile:130: all] Error 2

It seems, that libsmu tries to compile the python2 bindings even if told that it shouldn't.

https://stackoverflow.com/questions/64296359/how-can-i-install-pip-for-python2-7-in-ubuntu-20-04
After installing pip2.7 and installing cython with this old pip version i still need to install python-dev-is-python2
NOW everything compiles.

All alice scripts segfault with this installation. Pixelpulse2 segfaults too.

Is there a possibility to uninstall these files installed by make install?

meschi, 2021-10-04:
Okay, quick update:
My status after suspending/resuming the machine and plugging the adalm1000 out and plugging it in again:
I got pixelpulse2 running now, I tried running pixelpulse2 compiled against the .deb version of libsmu (which didn't work anyways) with libsmu built on my machine. Recompiling pixelpulse2 did the trick, it works now.

alice now seems to work, too.

The remaining problem is, that I cann't compile everything without the Python2.7 bindings. This causes me to install an old, hacky toolchain that isn't supported by Ubuntu 20.04

Is the ADALM1000 toolchain actually usable without python2.7? Is it fully ported to python3?

AlexandraTrifan, 2021-10-04:
Hi,

Libsmu is compatible with python3. The "-DUSE_PYTHON2=OFF" flag should have forced the build process to use python3. 

I will investigate why is this happening. In the meantime, could you tell us if you tried to compile initially without the "-DUSE_PYTHON2=OFF" flag and then added it? If the answer is yes, did you try to clean up the build folder and rerun the build process from the beginning?

Thank you!
-Alexandra

meschi, 2021-10-04:
Yes, I did a `make clean` between the builds. I've added a print statement to setup.py to verify via sys.version that it actually runs python2 and it does.

My build steps:
```
make clean
cmake .. -DBUILD_PYTHON=ON -DUSE_PYTHON2=OFF
make
```

meschi, 2021-10-04:
PYTHON_EXECUTABLE in the cmake-files evaluates to python2.7, 
```
grep -rnw . -e 'PYTHON_EXECUTABLE'
./bindings/python/CMakeLists.txt:16:		COMMAND ${PYTHON_EXECUTABLE} ${SETUP_PY} build build_ext -L "${CMAKE_BINARY_DIR}/src"
./bindings/python/CMakeLists.txt:22:		COMMAND ${PYTHON_EXECUTABLE} ${SETUP_PY} install --root=\$ENV{DESTDIR}/ --prefix=${CMAKE_INSTALL_PREFIX})")
./build/CMakeCache.txt:318:PYTHON_EXECUTABLE:FILEPATH=/usr/bin/python2.7
./build/CMakeCache.txt:617://ADVANCED property for variable: PYTHON_EXECUTABLE
./build/CMakeCache.txt:618:PYTHON_EXECUTABLE-ADVANCED:INTERNAL=1
```

https://stackoverflow.com/questions/15291500/i-have-2-versions-of-python-installed-but-cmake-is-using-older-version-how-do

bindings/python/CMakeLists.txt seems to use a deprecated method to select the python version. I'll see if I can fix that.

https://cmake.org/cmake/help/latest/module/FindPythonInterp.html

meschi, 2021-10-04:
Ok, I found the problem. I had been using an old CMakeCache.txt. Deleting this file solved my problem.
Building with `cmake .. -DBUILD_PYTHON=ON -DUSE_PYTHON2=OFF` properly compiles with python3. However `cmake .. -DBUILD_PYTHON=ON` only builds python2.7 bindings, not python3. Is this expected behaviour?

AlexandraTrifan, 2021-10-04:
It is expected. The default behaviour for libsmu is to look for Python2. 
But we will change this since Python2 is deprecated. After applying these changes, the "-DUSE_PYTHON2=OFF" flag will no longer be required. 

-Alexandra

meschi, 2021-10-04:
Thanks for the quick response, and thanks for the software :)

closed 2021-10-04 by meschi