analogdevicesinc/libsmu #192: Libsmu for python 3.8 in Win 10 OS

vsethia-adi, 2021-11-28:
Is there a libsmu package for ADALM1K in Python 3.8 in Win 10 OS?

Reason is libm2k for ADALM2K is available in python 3.8.

tfcollins, 2021-11-29:
The conda package should support 3.8

vsethia-adi, 2021-11-29:
I use Python pip package and not conda package.
This is for Win OS 10.
I downloaded the installer available on the site and it tries to install python 3.7 bindings. Given that python 3.7 is not installed, the bindings are not copied.
However, the libm2k is installed for python 3.8 bindings.

vsethia-adi, 2021-11-29:
@damercer  mentioned in EZ that python 3.8 bindings of libsmu exist in his Win 10 computer. But the same is not available on the libsmu Win installer. I am using the Python 3.8 environment from the python.org and not the Anaconda version.

damercer, 2021-11-29:
I installed the Python 3.8 bindings manually. I don't recall the exact steps but it was more or less copying the pysmu source files into site-packages and running the setup.py installer. Or something like that.

AlexandraTrifan, 2021-12-03:
Hi,

We need to work on some updates for our continuous integration builds. After the updates, there will be wheel packages provided for multiple Python versions (can be installed using pip). 

The easiest method right now is to use the Conda packages, as mentioned above. This will install both the libsmu library and the Python bindings in a separate Python environment. (https://anaconda.org/conda-forge/libsmu)
If Conda is not an option, I can prepare and get back to you with a short guide to build the Python bindings manually.

Thank you!
-Alexandra

vsethia-adi, 2021-12-05:
Hi Alexandra,
Do provide me with a short guide to build python bindings manually. If that did not work out, is there any other option other than Conda to use the libsmu package.

AlexandraTrifan, 2022-02-11:
Hi,

In the last weeks we worked on improving the build setup for libsmu and pysmu since it was a bit outdated.
You can try the latest release, which will make it much easier to install pysmu with the chosen Python version. (https://github.com/analogdevicesinc/libsmu/releases/tag/v1.0.4 )

This one **does not include the Python bindings** in the official release, so you just download the libsmu setup file and install the base library + the smu command line utility. 
As for the Python version, you can download the **libsmu-1.0.4-win32-win64.zip** archive which contains the build folders for both 64 and 32 bit. In the appropriate folder, you will find a pysmu-cp38-cp38...whl file. You can install that one using pip, just be sure to upgrade your pip first.
`python.exe -m pip install --upgrade pip`
`python.exe -m pip install pysmu-cp38-cp38...whl` (just put the chosen filename)
Or you can just use the following command, which does not require you to pre-download any .whl file (it just downloads it from the official Test PyPi repository)
`pip install -i https://test.pypi.org/simple/ pysmu` 
This one will detect your Python version and choose the specific wheel(pip package) for it.

 Thank you! 
 -Alexandra