analogdevicesinc/libsmu #193: Session.run() hangs with sample rates < 100kSps and number of samples > 256/512

snoehmer, 2021-11-29:
Hi,
first of all, thanks for all the work on libsmu and pysmu - I am currently trying to create some current pulse profiles with the ADALM1000 with some custom Python code using pysmu, and I ran into some issues with low sample rates.
The current pulses I want to generate are in the length of seconds (e.g. 10mA for 2s, then 5s pause), so the default sample rate of 100kSps is a bit overkill and leads to huge sample lengths. But when I reduce the sample rate to 10kSps or 1kSps, libsmu/pysmu seems to get stuck in an endless loop when calling Session.run() (or Session.start(), or Device.get_samples(), etc.).
The interesting thing is that everything works as expected as long as the number of samples requested is below a certain threshold - which seems to be always a power of 2. But this only applies to sample rates below 100kSps (more specific: below 10240Sps).

This behavior can be triggered with very simple code:

```
import pysmu

samplerate = 1000
n_samples = 257

s = pysmu.Session(sample_rate = samplerate)

s.run(n_samples)
samples = s.read(n_samples)
```

The above example hangs at `s.run(n_samples)`. It seems the same happens for all values `n_samples > 256`.
If I change the code to `n_samples = 256`, the code works. Again, it seems it also works for all values `n_samples <= 256`.

The same happens if I change the code to `samplerate = 10000`, but with the next power of 2 for the number of samples.
With `n_samples = 512` (and below) the code works, with  `n_samples = 513` (and above), the code stops working.

Since the sample rate is limited to 1kSps on the lower end by libsmu (as is seems), I cannot check further. Also, when I increase the sample rate to `samplerate = 100000`, the code does work regardless of `n_samples` (I tried up to 10M samples, along with Session.queue_size set to 10M).

I played around a bit with the sample rate, and saw that this behavior seems to happen only with `samplerate <= 10240` (but works fine for 10241 and above), with a threshold of `n_samples > 512`, and with `samplerate < 5120` and threshold `n_samples > 256`, which is oddly specific.

Can you reproduce this behavior? I am using libsmu 1.0.3 from conda (installed the latest release today) with Python 3.9, but I had the same issue with Python 3.7. The ADALM1000 is use is HW revision F with FW version 2.17.
Please let me know if I should provide any further information.

Thanks,
BR
Stefan

AlexandraTrifan, 2021-12-03:
Hi,

Unfortunately I didn't manage to reproduce the issue you experienced. I ran the example code you provided, tested a few combinations of nb_samples versus samplerate (similar with what you described), but the script did not hang.
I ran the tests on a Windows 10 machine, libsmu installed today using the conda packages, on a Python 3.7 64bit env, with the ADALM1000 rev F with firmware 2.17.

Could you tell us what operating system are you using? 
Also, could you check if there are any other libsmu/pysmu versions installed on your machine (older versions) that may somehow affect or interact with the version you are running?
Could you run "conda list" and send us the build version for libsmu?

Thank you!
-Alexandra

snoehmer, 2022-02-03:
Hi Alexandra,

thanks for the quick reply, and sorry for the late response. I unfortunately didn't find the time to dig into this deeper until now.

To answer your questions first:
The operating system is Windows 10 64bit, running Python 3.9 (64bit)
Yes, there were other libsmu/pysmu installations, from the latest Windows installer on the Github releases page (see below for more info)
The `conda list` build info is `libsmu   1.0.3   py39h325aaa3_2   conda-forge`

After trying a few things, I found out that the script also worked on another Win10 machine - which also had Python 3.7 installed, but with a fresh install of libsmu/pysmu from conda (no previous libsmu versions were installed). After setting up a Python 3.9 environment on that machine, and installing libsmu from conda, I again got the same results as on my main machine - a hang when I run the script. Tried on a 3rd machine, same results. So it seems the issues is the Python version.
Interesting thing: with Python 3.8, the script also hangs. However, with Python 3.10 it runs normally. 

Going back to the original machine, I uninstalled the previous libsmu installation (the Github releases Windows installer one) as per your suggestion, then created a new Python 3.7 virtualenv and installed libsmu from conda - and it worked.
Tried it with a new, clean Python 3.8 virtualenv - didn't work.
Same with Python 3,9 - didn't work.
Same with Python 3.10 - works.

So it seems this behavior depends on the Python version. By the way, I am using Anaconda to manage the Python versions, packages and virtualenvs.
With the correct Python version, I can now continue working with the ADALM1000.

If you need any further information, please let me know.

Thanks!
BR
Stefan

AlexandraTrifan, 2022-02-07:
Hi,

I ran some tests on all the python versions you mentioned, using Anaconda to manage them, and I can confirm the issue. 
I am trying to track it down and get a fix for it.

Thank you!
-Alexandra

snoehmer, 2022-02-07:
Hi,

thanks for the quick response. Please let me know if I can do anything to help.

BR
Stefan