analogdevicesinc/libsmu #194: Make error, no module named setuptools

SuperSecureHuman, 2022-01-05:
I am trying to install, but it says module error. I can confirm that I have installed setuptools.

[image]

System info

[image]

Thanks!

AlexandraTrifan, 2022-01-05:
Hi,

Can you send us the output log from the CMake step? This way we can check if cmake linked to the correct python version, the one in miniconda/envs/python37.

-Alexandra

SuperSecureHuman, 2022-01-05:
[CMakeOutput.log](https://github.com/analogdevicesinc/libsmu/files/7813007/CMakeOutput.log)

Here is the log file

I am not exactly sure about cmake being linked. Can you link an article or something?

Thanks!

AlexandraTrifan, 2022-01-05:
Please also attach the CMakeCache.txt and a printscreen from the command line, from when you ran the step "cmake ..", right before "make".
When "cmake .." is called, the project searches for some dependent libraries. Python is one of those dependencies, and CMake will find the one that's in the system Path. 

-Alexandra

SuperSecureHuman, 2022-01-05:
cmake .. 

![image](https://user-images.githubusercontent.com/88489071/148186667-c12f6920-9dd7-49c0-a051-2279985dc496.pn

Cmake Cache
[CMakeCache.txt](https://github.com/analogdevicesinc/libsmu/files/7813324/CMakeCache.txt)
g)

Here you go

AlexandraTrifan, 2022-01-06:
Hi,

The problem is that CMake detects python2.7 before python3 and uses that one. Please remove everything from your "build" directory and try to build it again in the following way: 

> cmake -DUSE_PYTHON2=OFF ..
make

-Alexandra

SuperSecureHuman, 2022-01-06:
Yep!
It did the trick!

Thanks
You might wanna update readme (the table where you put flags)

I'll close this issue now :)

closed 2022-01-06 by SuperSecureHuman

AlexandraTrifan, 2022-01-06:
We are working on some updates and this flag will disappear in the future.

-Alexandra