analogdevicesinc/libsmu #200: Begin() Read Violation Null Pointer

elwool, 2022-03-08:
I'm currently trying to demo an Adalm1000 but running into issues with libsmu

Unfortunately I'm not a C++ expert. I haven't tried the python version yet - 
I was hoping to be able to drive this unit from LabView but the library isn't written in a way that I can do this directly.
So now I'm just trying to get the examples to work.

The examples all fail during begin() - read violation null ptr
[image]

Is there anything obvious I should check?

AlexandraTrifan, 2022-03-08:
Hi,

Could you provide a more detailed description on what you are trying to do?
I can see there is a Source.cpp which I believe it contains an example for libsmu. (please correct me if I am wrong). If so, could you provide the example you are running?

You did manage to build the example and it fails at runtime?
Did you manage to correctly link the example executable to libsmu?

Is the libsmu library located somewhere that is accessible from the Environment Path? It's a common process on Visual Studio builds to copy the library (libsmu.dll in this case) in the same location as your example.exe (otherwise, the executable might not find the library at runtime).  

Thank you!
-Alexandra

elwool, 2022-03-08:
Hi Alexandra,

Source.cpp is a copy and paste example of one of the examples:
[image]

The example successfully builds, it fails at runtime.
What i did notice was that the watching the session variable shows
- m_available_devices {size=0}
- m_active_devices 1

I have placed libsmu.dll in the same directory as the example.exe

AlexandraTrifan, 2022-03-08:
(I think libusb.dll should also be placed near de libsmu.dll)
Could you provide some details on the machine you are using? OS version, 32/64bit?
We will also give it a try locally on a Windows machine to check all the build steps and figure out what might be missing.

-Alexandra

elwool, 2022-03-08:
Hi Alexandra,

I also put libusb.dll into the same folder

This is running Win10 21H1 64Bit
Using libsmu-1.0.4-setup-x86.exe

Thank you!
- Liam

AlexandraTrifan, 2022-03-09:
Hi,

Can you try to build in Release mode, rather than Debug, and see if that fixes your issue?

-Alexandra

elwool, 2022-03-10:
Hi Alexandra,

I've just tried building in release and it works. 

Oops! I possibly used release DLLs when trying to perform a debug build 🤦 

Thanks Very Much for your help, it's greatly appreciated! 
Liam

closed 2022-03-28 by AlexandraTrifan