analogdevicesinc/libsmu #201: Python demo code - continuous read/write failure

chamoujacon, 2022-04-27:
Windows 10, using Python 3.10, REV-F board. Installed using "pip install -i https://test.pypi.org/simple/pysmu". Code runs/prints for a few seconds then throws an error.

Traceback (most recent call last):
  File "libsmu.pyx", line 626, in pysmu.libsmu.SessionDevice.write
RuntimeError: data sample dropped: The directory cannot be removed.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\XXX", line 67, in <module>
    chan_a.write(refill_data(num_samples, i % 6), -1)
  File "libsmu.pyx", line 782, in pysmu.libsmu.Channel.write
  File "libsmu.pyx", line 636, in pysmu.libsmu.SessionDevice.write
pysmu.exceptions.SampleDrop: data sample dropped: The directory cannot be removed.

AlexandraTrifan, 2022-05-02:
Hi,

I ran a test on a clean Windows 10 PC, using the same setup you described above and I can't reproduce the issue.
Could you provide some more details? 
Are you running the code from the Command Prompt or some different command line?
Did you have an old libsmu version installed before installing this one (v0.4.0)? 

Thank you!
-Alexandra

chamoujacon, 2022-05-04:
Hi Alexandra,

Thanks for the message! Please find answers below:

Are you running the code from the Command Prompt or some different command line?
**I'm using the Pycharm IDE. After more testing, I found that the code will run without errors if either "Emulate terminal in output console" or "Run with Python Console" is checked**. If neither is checked, it will run for a few seconds then throw the error above.
[image]

Did you have an old libsmu version installed before installing this one (v0.4.0)?
**No, this was a fresh install of libsmu**

johthu, 2024-03-24:
I'm having the same issue. I don't really understand how the buffer system works, but I'm using 

```python

def refill_data(num_samples, voltage):
    return [voltage] * num_samples

num_samples = session.queue_size + 1
chan_a.write(refill_data(num_samples, voltage_out[0]), -1)

``` 
I'm getting the error
```
Traceback (most recent call last):
  File "libsmu.pyx", line 626, in pysmu.libsmu.SessionDevice.write
RuntimeError: data write timeout, no available queue space: The directory cannot be removed.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):

File "...", line 227, in run_mppt
    chan_b.write(refill_data(num_samples, voltage_out[1]), -1)
  File "libsmu.pyx", line 782, in pysmu.libsmu.Channel.write
  File "libsmu.pyx", line 638, in pysmu.libsmu.SessionDevice.write
pysmu.exceptions.WriteTimeout: data write timeout, no available queue space: The directory cannot be removed.
``` 
and
```
File "libsmu.pyx", line 626, in pysmu.libsmu.SessionDevice.write
RuntimeError: data sample dropped: The directory cannot be removed.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\John\anaconda3\envs\python3.10_env2\SMU_Combined.py", line 226, in run_mppt
    chan_a.write(refill_data(num_samples, voltage_out[0]), -1)
  File "libsmu.pyx", line 782, in pysmu.libsmu.Channel.write
  File "libsmu.pyx", line 636, in pysmu.libsmu.SessionDevice.write
pysmu.exceptions.SampleDrop: data sample dropped: The directory cannot be removed.
``` 
The error appears after a few minutes at 100,000 samples/s, an hour at 10000, and a few hours at 6000. I am considering using channel.flush() to clear the buffers before each write? I have no idea how to get this to stop happening, it crashes the program, and this is being used in a study that is supposed to run for 500hrs. Appreciate the help,
Regards,
John