analogdevicesinc/libsmu #207: Version compatable with Mac computers with M1/M2 processors?

damercer, 2023-05-08:
Hi:
I've been contacted by a professor who is trying to use libsmu on Mac M1/M2 processor computer(s) and having trouble  getting it to work. Are you aware of any differences / issues when compiling libsmu targeting for M1/M2 processors? Has libsmu been tested on M1/M2 Macs? Any plan on offering versions specific for these Mac's?

Thanks
Doug

damercer, 2023-05-09:
Some addition information on this issue. It seems to be more related to the Python bindings of libsmu. I'm told that it was a common issue for the Intel vs M1 processor Mac versions early on with Python code, particularly anything which was related to Anaconda.

Doug

AlexandraTrifan, 2023-05-12:
Hi,

I checked the builds on M1/M2 processors. There are some small fixes that need to be done in order to be able to build the core library for the required architecture. I will commit those in the next few days.
As for the Conda packages to be updated and compatible with these processors, we need to create a new release. A new libsmu release would trigger a new conda and pip package. I will work on adding it, but I can't provide a specific timeline for this. 
The quick fix (until I manage to release a new package) would be to compile the library from sources, after I commit the above mentioned fixes. 

Thank you!
-Alexandra

damercer, 2023-05-12:
Thanks for looking into this. It probably explains some other un-specified MAC related complaints I've gotten over the last few months.
Doug.